PiSupply/IoTLoRaRange #3: iot-lora-status

ryanteck, 2018-12-18:
New program:
Python service that

- Checks for internet connection
- Checks if the packet forwarder is running
- Checks temperature of RPi
- Command can be run to view in terminal
- GPIO Output mode

ryanteck, 2018-12-18:
Moved to IoT LoRa Range,

New idea!

I'm thinking lets have a website that runs on the RPi, it provides a web interface on a port.

Basic HTTP Authentication,

Does the above plus we can integrate an interface similar to the iotloraconfigure website so that it allows for simple configuration of the gateway over a network.

This could hopefully make it even easier to setup but also allow the user to monitor their gateway easier.

@shawaj what do you think? A mix of the web config site plus a status monitor site hosted on the gateway itself.

shawaj, 2018-12-18:
Sounds ideal to me!

ryanteck, 2018-12-18:
Great!

No need to upload the configurator to the website then.

ryanteck, 2018-12-18:
Accidentally closed

ryanteck, 2018-12-26:
Work is continuing well. Will be in the 01-01-2019 Release of the SD Card image.

ryanteck, 2018-12-27:
This has required us to move the configuration files back onto the linux partition of the system. While this won't allow users to modify the configuration file using the SD card on their windows PC we shouldn't need to with the configurator.

ryanteck, 2018-12-27:
This tool also needs to be able to set the frequency of the concentrator.

closed 2019-07-27 by ryanteck