PiSupply/IoTLoRaRange #4: Upload images

ChristopherRush, 2019-05-15:
Add images here to link in repo

ChristopherRush, 2019-05-16:
[image]
[image]
[image]

ChristopherRush, 2019-05-16:
[image]

shawaj, 2019-05-16:
@ChristopherRush just FYI Louis will be taking new pictures of these soon

ChristopherRush, 2019-05-16:
@shawaj Perfect, will replace these ones when done.

ChristopherRush, 2019-05-17:
[image]

ChristopherRush, 2019-05-20:
[image]

monglet, 2019-05-21:
Need a new phat photo!

ChristopherRush, 2019-05-31:
[image]

closed 2019-08-02 by ryanteck

ChristopherRush, 2019-12-11:
[IP67 Case.pdf](https://github.com/PiSupply/IoTLoRaRange/files/3950846/IP67.Case.pdf)

shawaj, 2019-12-11:
@ChristopherRush just so you are aware this drawing is not completely correct...the holes are the wrong size. I think most of the internals are correct but not the hole bores through the side

shawaj, 2019-12-11:
@ChristopherRush  [image]

ryanteck, 2019-12-11:
@shawaj do we have all of the dimensions in Metric as some being imperial and some in metric doesn't make it clear? 

Do we have the metric value for the others?

ChristopherRush, 2019-12-11:
@shawaj Thanks I did see that other drawing you posted. A customer just wanted the internal dimensions. Do we have a final drawing for the case? We should post it somewhere more visible I guess. Maybe create a repo on the Nebra GitHub.

shawaj, 2019-12-11:
@ryanteck 5/8-24 is a particular thread type. It does not have a metric equivalent.

@ChristopherRush we do not have a final drawing - but you could just ammend the one you have with the comments on my picture and then it will be correct. Or ask @LouisReed to?

Once we have it, we can post it on Shopify and link to it on the product page and/or make a repo as you say

ryanteck, 2019-12-11:
@shawaj do we have a link to glands that will fit as I've been asked by customers what glands will fit but haven't been able to find any before and can't with the 5/8-24

shawaj, 2019-12-11:
@ryanteck 5/8-24 is for the bulkhead connectors, not glands

An M15 or M16 might work - want me to order one of each to you to check?

[image]

ryanteck, 2019-12-11:
If you could maybe, what it is is i've had some customers ask for the bulkhead connectors if there's glands instead for cables to be passed through instead if possible. 

Thanks

shawaj, 2019-12-11:
@ryanteck m15 doesn't seem to exist. Have ordered an M16 one from RS to you - it will probably be a little tight but since it is plastic going into metal it should work ok I reckon.

ChristopherRush, 2020-02-05:
[image]

ChristopherRush, 2020-02-06:
[image]