PiSupply/IoTLoRaRange #6: Use Ubuntu Server / Core for Standalone Gateway

ryanteck, 2019-07-01:
This is to discuss with Beta Testers & Others the software for the standalone gateway.

With the release of Debian Buster a majority of bits did break and while they are fixed now I have noticed quite a few issues.

As stability is the key concern for this I have investigated the potential for switching to Ubuntu.

Primarily as a spur on from issue https://github.com/PiSupply/iot-pi-gen/issues/2 . The two 4G Modules we have didn't seem to want to work with raspbian or kept having intermitternt issues with connection.

Trying with Ubuntu Server worked instantly once the packages were installed.

### Test List
This all has to be tested before a switch can be made.

- Packet Forwarder [ ] 
- Nginx  [ ] 
- PHP  [ ] 
- Controller  [ ] 
- Server  [ ] 

### Pros & Cons
Initially the main Con would be that it doesn't support Original Raspberry Pi's. However as this is primarily for the standalone gateway this won't be an issue as I believe we're planning on shipping with CM3+ 8GBs.
Along with this there might be slightly more work to setup Ubuntu to Compile. 

However just at initial lookings there's many benefits of moving to Ubuntu Server including:

- Ubuntu 18.04 has LTS support until 2023
- Because of it's LTS support software has significantly less chance of breaking. 
- 64 Bit Arm because its 2019. Might also help significantly speed up compiling times and less chance for bugs as caused by the buster upgrade.

###Note
We will still be providing a template for balena however most customers that are using balena will likely be using their own images / templates. This image will be pre-flashed onto all the CM3+s leaving the factory for those wanting to buy something they can just login, copy keys and get going.

ryanteck, 2019-07-01:
The EC25-E and SIM7100 both worked straight away with Ubuntu. Configuration just required one command to set the sim provider.

monglet, 2019-07-01:
Ubuntu server and the 40pin breakout. How's that going to go, in general? Presumably most GPIO stuff is fairly portable across?

shawaj, 2019-07-01:
@monglet that was my first thought too - whether it would break support for existing hats

However, as you say @ryanteck the biggest concern is stability of the core components. If hats are a bit more fiddly it's not the end of the world IMO. Stability should take precedence

AmedeeBulle, 2019-07-01:
Is there any statement on term of support for Raspbian Stretch?
From a pure Debian perspective, Buster is still development and Stretch the current stable release which will be supported until June 2022...

Although I believe 64bits is the way to go these days, I had variable success with mainline kernels on Raspberry Pi when it comes to "hardware management" (GPIO, CPU throttling, ...), so I tend to stick to the _official_ Raspbian on my Pies...

ryanteck, 2019-07-01:
@shawaj I think we can do a check with the most important hats to us.

I don't see Pijuice not working and I can test that out tonight / tomorrow

@AmedeeBulle That's quite a good point actually. I've just done some googling.

It looks like packages are supported for the same time as in the main debian repositorys. However it looks like kernels and some RPi supported packages are not maintained once they switch. As RPi are now shipping buster it doesn't look like stretch will get any updates out of them. So Ubuntu would be a good move for the better software support.

shawaj, 2019-07-01:
@ryanteck yeah, whatever you think is best really. I think it's probably easier to bring support for X HAT to a stable platform than to bring stability. If that makes sense

monglet, 2019-07-01:
Thinking through HAT compatibility. The standalone is presumably going to spend most of it's time in the ip67 case. PiJuice would be excellent for UPS behaviour. i2c hats would be easier than SPI on ubuntu. Wondering about display HATs too, for stats display. (sorry, typing my thoughts out loud!)

ryanteck, 2019-07-01:
@monglet SPI and I2C both seem to be fully working under Ubuntu as standard as these just show as linux devices. It'll be more if there's reliance on dtbs and gpio libraries.

Already got the gateway hat working with a 3B+ with no modifications required to the software. Just need to re-test on CM3+

I'll test pijuice tomorrow however expect no issues.

ryanteck, 2019-07-02:
All Lora Gateway Hat Tests completed, 100% Working.

Just need to test on CM3+.

AmedeeBulle, 2019-07-02:
2 Qs:
- Do you have a working RPi.GPIO lib (to reset the concentrator), or are you using /sys?
- Is there any CPU throttling and can you disable it? (to avoid issues with SPI)

ryanteck, 2019-07-02:
To reset the concentrator we've been using /sys without issue. The PHP controller calls the bash script to do this on reboot.

As for throttling  I haven't investigated. I suspect it is the same as Raspbian where it will throttle with heat.

I haven't actually had to set the GPU Clock slower which I have set as default in the Raspbian image. However I also have the packet forwarder use a slightly slower SPI speed for reliability. (To ensure less issues across all devices). 

I'll test to see if RPi.GPIO works.

ryanteck, 2019-07-02:
Pi Juice works with little modification on Ubuntu. The only thing was a permissions error on a command but with chmod easily fixed that.

However RPi.GPIO doesn't seem to want to work. I'll see what steps are required to get it to work.

[image]

Otherwise all tests are indicating Ubuntu is the better option so far.

shawaj, 2019-07-02:
Sounds good!

AmedeeBulle, 2019-07-02:
RPi.GPIO is not necessary -- `/sys` is good enough...
(I have somewhere a python library which emulates the basic functionalities of RPi.GPIO, but that's not a even useful here)

This project is more about having an (open) appliance, people who want to do more are better with a plain RPi / LoRaWan hat...

ryanteck, 2019-07-02:
Yeah, I believe it'll work but it's just whatever signature it's looking for to confirm it's running on a RPi is failing. 

So far I think it's all confirming a change to Ubuntu for the Standalone Gateway. The current image for HATs will stay as it is along with being actively maintained and updated. 

The main thing left is to investigate how to create our own ubuntu images and confirm if we need to use Core or if we can use Server

ryanteck, 2019-07-06:
### New Update

So Ubuntu Core might not be possible but is waiting on clarification to confirm. As essentially we would need to pre-load the software / snaps into our own Ubuntu Image this may not be then allowed to be re-distributed without paying licensing fees. However I am waiting on clarification. 

Also when setting up on  my CM3+ I found that the Ubuntu Core & Ubuntu Server Images don't have the latest firmware for the Raspberry Pi so it refused to boot, however while I could re-package the boot files using a guide Ubuntu provide for us to then include in our image. This could count as a change which once again falls into the licensing fee category above.

For this project from talking to a few people having some kind of method to ensure reliability and such is key, primarily the ability to remotely update the gateways.

One other potential route I'm investigating is Open Balena which would allow us to setup our own Basic Version of Balena. Each gateway would get programmed with the OS image and then when powered on it would contact a server that we would setup to download and configure the gateway. 

The main downside is that we would need to setup our own server for the items to download from but otherwise could actually be an even better route than Ubuntu

ryanteck, 2019-07-08:
I'll get the balena option tested out this week if we don't have clarification from Canonical with the issues we've got of using Ubuntu Core

ryanteck, 2019-07-12:
Thank you to all those who have provided feedback.

I think it will actually be best to try to stick to Raspbian so will work on fixing the 4G Module bugs with that. This is primarily as much to avoid duplicating lots of work.

However it will be where we look at packaging our code into basic APT packages on potentially our own APT repository. That way gateways can be updated with a one click button instead.

Support will still be given for people wanting to use normal Balena.

closed 2019-07-12 by ryanteck