PiSupply/IoTLoRaRange #7: Raspberry Pi 'Internet Check' WebPage

adamboutcher, 2019-07-02:
It would be advantageous for you do a better network check...

`$internetCheck1 = file_get_contents('http://www.google.com');
$internetCheck2 = file_get_contents('https://status.thethings.network/');
$internetCheck3 = file_get_contents('https://www.cloudflare.com/');`

You could check for the local gateway and maybe 1.1.1.1 / 8.8.8.8 which would let you output LAN Status, WAN Status and DNS Status.

jezmck, 2019-07-02:
Is this on the correct repo?

ryanteck, 2019-07-02:
I admit that code isn't quite the best however it actually works surprisingly well.

By looking up actual websites it caught SSL issues during supporting Buster and essentially does confirm the DNS lookups are working. 

I'll consider alternate routes. The IP address lookup itself isn't working so parsing "ip a" or something might be a better route to go. However I'll likely still use the old method for one part of this.