PiSupply/IoTLoRaRange #12: Orange PC +

coraxyn, 2020-04-08:
Using Orange PC Plus, Armbian and LoRa pHat.  Connecting via UART3 and /dev/ttyS3

Connecting and configuring UART seems to work, writing to UART may be working, but cannot get anything back from LoRa (via UART)
Any recommendations?

ryanteck, 2020-04-08:
Hi @coraxyn ,

so to confirm that you're able to confirm you've configured UART correctly under Armbian and can confirm that it's working but it's just the LoRa pHAT not working? 

Have you toggled IO on the GPIO pin used for reset? This would be Pin 11 or PA1

To reset you would pull the pin low then high

coraxyn, 2020-04-08:
Will try this.  Thank you

ryanteck, 2020-04-08:
The baudrate should also be 115200 as default.

coraxyn, 2020-04-09:
Tried as you suggested, but nothing.

Looking at various python, c, and pascal code - all read or write to
something near /sys/class/gpio .. something.  Looking at file system the
specified file(s) in gpio directory do not exist.   Am running
armbian/buster.   Can you help?  Am totally lost

Paul

coraxyn, 2020-04-09:
After further reading found proper 'files' to use.  Thank you

ryanteck, 2020-04-09:
Hi Paul, I assume this means it's all working now and I can close this issue?

coraxyn, 2020-04-09:
Yep

closed 2020-04-09 by ryanteck