PiSupply/IoTLoRaRange #14: Upgrade firmware of RAK811 modules

ryanteck, 2020-05-12:
Github thread to track firmware upgrade process for the RAK811 modules.

Any customers wanting newer firmware should comment or +1 this thread so we can keep track.

panban-ux, 2020-11-03:
How to update the firmware on this lora pHat sitting atop a Pi 4 in raspian, can someone elaborate?

ryanteck, 2020-11-03:
Currently it is not possible @panban-ux because it breaks a majority of the code currently on it.

panban-ux, 2020-11-03:
Thanks for the reply Ryanteck. According to cutecom i got the [21:37:38:053] OK2.0.3.0␍␊ firmware. So the version found out here RAK811_HF_V3.0.0.13.T3 is not aligning with the code of using the pyrak811 library.
 Sorry on the questions im just new to the matter and try to initialise my board and registration it to TTN

ryanteck, 2020-11-03:
The library is only designed for the 2.0.3.0 firmware currently.

Unfortunately the firmware had a big re-design which means that it is no longer compatible with the current version of the library which will then need updating. However at the same time the last time I tried a firmware upgrade the process had changed causing delays.

panban-ux, 2020-11-03:
Thanks man, i'll be keep my self notified on the matter for any future library update.

EdwardHanlon30, 2021-03-10:
Im just curious how these boards where tested after production if they all went out and none of them work . Has this happened before ? What type of time delay are we looking at here ? 
Please let me know asap when this is resolved I am on a tight timeline

ryanteck, 2021-03-10:
Hi @EdwardHanlon30 & @markus1203 ,

I've opened a new issue about the issues at https://github.com/PiSupply/IoTLoRaRange/issues/31