PiSupply/IoTLoRaRange #15: Not sending data

DevScienceScope, 2020-05-26:
Hi I have been playing with 2 of the mirco:bit LoRa Nodes and having relibility issues where the data is not being sent or recieved. I have an arduino MKR 1300 and 1310 devices running in the same location to a gateway in the same room working >99% for 4 days now. so i dont think it is an issue with the gateway or signal strength. 

To get the LoRa nodes working i have to keep resetting them until they start working. Sometimes they need to be left on for about 10 minutes before the data is eventually sent.

ryanteck, 2020-05-26:
Hi @DevScienceScope ,

Do you have screenshots of the code you used at all?

What region are you in as well as this can help.

DevScienceScope, 2020-05-26:
Im in the UK so i set it to EU868
[image]
[image]

ryanteck, 2020-05-26:
Hmm, so every time you press the button it doesn't work or are you looping it? 

Also to just check how is the micro:bit being powered.

DevScienceScope, 2020-05-26:
i have just being trying to test it with button press. I try a button press every 10 seconds for several attempts then i reset and try again. sometimes it works and sometimes it doesnt. I always get the LoRa Ready message and wait for it to finish
I am powering it via the computer using USB cable

shawaj, 2020-05-26:
@DevScienceScope is it just that the gateway isn't receiving the messages?

If so, have you checked that you have unchecked the "Frame Counter Checks" option in the device settings?

DevScienceScope, 2020-05-27:
I have unchecked the Frame Counter Checks and it appears to be working now. i have restarted the device a couple of times now and it is sending data each time. what could be causing this issue as i belive its important to have Frame Counter Checks on

ryanteck, 2020-05-27:
The frame counter is used to ensure that the transmitted packets match what's expected. Each time a packet it sent the number of packets transmitted that is included in the data is increased (so if the nodes transmitted 10 times the counter will be at 10).

When you're resetting the microbit this restarts the counter on the node back to 0 but then because the counter on TTN is at 10 it'll ignore all packets before this. So essentially with it enabled you'd have to reset the counter on TTN every time you reset the node.

It's done as a bit of a security measure as if another person gets hold of your keys they can start transmitting but if your node is already at packet 100 and counting, they'd have to transmit to at least then for it to be accepted.

The other solution is to use OTAA mode but I can't remember what stage of implementation I got to with this, as at each reset that re-negotiates the keys and counters. 

However for the micro:bit node the need for OTAA isn't as high and neither is worrying with frame counters (as it'd be rare for an application needing top security using a micro:bit).

However as the issue is fixed I'm going to close this issue.

closed 2020-05-27 by ryanteck

DevScienceScope, 2020-05-27:
Thank you for the explanination. I'll keep an eye out for when the OTAA gets implemented. Thank you for your help and the quick response