PiSupply/IoTLoRaRange #16: Inducer vs bridge

AvdN, 2020-08-08:
To hook up the external antenna to the micro:bit node hat, I desoldered the inductor on the INT and managed to make it disappear.
The description indicates I can re-use the inductor on EXT or bridge. Is there no negative side-effect for doing the latter?

If I want to go back to the on board antenna, can a bridge be used as well?
What would be the specifications for the inductor if I want to order them somewhere?

ryanteck, 2020-08-08:
Hi @AvdN ,

A bridge won't work quite as well as if you have the inductor on it, however in the past I've ran it with just a bridge myself. 

Officially the inductor method is what we recommend.

The component for the inductor is LQP03TN18NH02D and should be available from many electrical component suppliers such as Farnell, Digikey, Mouser etc.

AvdN, 2020-08-09:
Thanks Ryan.

So far I found 9¢ for the part, €18 for the shipping ;-) Maybe I better buy a new board.
The part seem to have polarity, does the orientation matter (and if so, what should it be)?

The bridge seems to work, unfortunately my closest gateway connection point is a bridge 1-2 km from my home. Not ideal for a short development cycle.

ryanteck, 2020-08-09:
Hi @AvdN ,

For this application the inductance doesn't matter.

But indeed unfortunately small quantities from suppliers can hit high shipping fees. I tried looking for an alternative inductor but couldn't find any which would be cheaper.

AvdN, 2020-08-10:
I ordered the Pi Supply gateway directly from Pi Supply just a few minutes ago. That should make testing/development more easy.

If you you happen to have a spare inductor and assuming you are near the shipping facilities of Pi Supply, maybe you can include one

ryanteck, 2020-08-10:
Hi @AvdN 

Sorry but I work from home (even before covid) and the warehouse is around 150 miles away from where I live and I don't think anyone in the warehouse has one sorry.

AvdN, 2020-08-10:
No problem, I just thought I ask, just in case.

PR #17 (2020-08-10): add part number to README
Just in case someone is as smart as I was, and catapult the desoldered item from the board and make it disappear.
Closes #16
closed 2020-08-10 by ryanteck