PiSupply/IoTLoRaRange #19: Version of calibration firmware not expected

wkirby007, 2020-10-02:
Hi again, getting an error running the packet forwarder as below - using ttn in the EU

11:10:17  INFO: [Transports] Initializing protocol for 1 servers
11:10:17  INFO: [TTN] server "bridge.eu.thethings.network" connected
11:10:17  INFO: [main] Starting the concentrator
ERROR: Version of calibration firmware not expected, actual:104 expected:2
11:10:17  ERROR: [main] failed to start the concentrator

ryanteck, 2020-10-02:
This is most commonly caused by not running the reset script first, can you confirm that you are running the reset script before the packet forwarder?

wkirby007, 2020-10-02:
Hi, I run the reset program, which displays "root" and takes about 4-5 seconds to execute. I then run the packet forwarder and get the below.

pi@EV1PLGW01:/opt/iotloragateway $ sudo ./reset-22.sh
root
pi@EV1PLGW01:/opt/iotloragateway $ ./mp_pkt_fwd
13:14:36  *** Multi Protocol Packet Forwarder for Lora Gateway ***
Version: 3.0.20
13:14:36  *** Lora concentrator HAL library version info ***
Version: 5.0.1; Options: native;
***
13:14:36  INFO: Little endian host
13:14:36  INFO: found local configuration file /opt/iotloragateway/local_conf.json, parsing it
13:14:36  INFO: /opt/iotloragateway/local_conf.json does not contain a JSON object named SX1301_conf
13:14:36  INFO: /opt/iotloragateway/local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
13:14:36  INFO: gateway MAC address is configured to 000000000000000E
13:14:36  INFO: Found 1 servers in array.
13:14:36  INFO: Server 0 configured to "bridge.eu.thethings.network"
13:14:36  INFO: packets received with a valid CRC will be forwarded
13:14:36  INFO: packets received with a CRC error will NOT be forwarded
13:14:36  INFO: packets received with no CRC will NOT be forwarded
13:14:36  INFO: GPS serial port path is configured to "/dev/serial1"
13:14:36  INFO: Reference latitude is configured to 0.000000 deg
13:14:36  INFO: Reference longitude is configured to 0.000000 deg
13:14:36  INFO: Reference altitude is configured to 0 meters
13:14:36  INFO: GPS is disabled
13:14:36  INFO: Upstream data is enabled
13:14:36  INFO: Downstream data is enabled
13:14:36  INFO: Ghoststream data is disabled
13:14:36  INFO: Radiostream data is enabled
13:14:36  INFO: Statusstream data is enabled
13:14:36  INFO: Beacon is disabled
13:14:36  INFO: Packet logger is disabled
13:14:36  INFO: Flush output after statistic is disabled
13:14:36  INFO: Flush after each line of output is disabled
13:14:36  INFO: Watchdog is disabled
13:14:36  INFO: Contact email configured to "xxx"
13:14:36  INFO: Description configured to "xxx"
13:14:36  INFO: [Transports] Initializing protocol for 1 servers
13:14:36  INFO: [TTN] server "bridge.eu.thethings.network" connected
13:14:36  INFO: [main] Starting the concentrator
ERROR: Version of calibration firmware not expected, actual:104 expected:2
13:14:36  ERROR: [main] failed to start the concentrator

ryanteck, 2020-10-02:
Just to check, do you have a global_conf.json file?

wkirby007, 2020-10-02:
Thanks, I must have added a "." when applying the rename following the downloading. All working now.

Thanks for your prompt response.

ryanteck, 2020-10-02:
Excellent, good to hear that it's fixed and fully working.

closed 2020-10-02 by ryanteck