PiSupply/IoTLoRaRange #20: GPS Read Fail

wkirby007, 2020-10-05:
Hi, the gateway is up but I'm getting a continuous error since adding the GPS module.

WARNING: gps read() returned value 0

I has similar issues using another GPS module on a Pi4 where the BLE onboard was using serial0 by default. I get this error when using serial1. If I use serial0 as instructed, I get a warning during initialisation of the packet forwarded stating the below.

WARNING: [main] impossible to open /dev/serial0 for GPS sync (check permissions)

Followed by this.

INFO: Enabling GPS mode for concentrator's counter.

And then this.

### [GPS] ###
# Invalid gps time reference (age: 1601910139 sec)
# no valid GPS coordinates available yet

I've used the connected GPS antenna on another build using a different GPS module and get a fix as expected.

ryanteck, 2020-10-05:
Hi Will,

Have you disabled the BLE chipset?

wkirby007, 2020-10-05:
Hi Ryan,

I've fixed it now, I had to enable the serial interface for it to work - its reporting correctly now.

Does the GPS module feed the location of the gateway in TTN? Not sure what the benefit of having it onboard is, are there any resources you can recommend?

ryanteck, 2020-10-05:
Excellent, glad it's fixed.

As to weather GPS is beneficial is slightly up for debate still in my view.

It feeds the location of the gateway to TTN, which in some cases can be beneficial to those who have gateways they take around to different locations. But as most gateways are fixed this isn't really a benefit to most.

Technically the main benefit is the time provided by a GPS module is more accurate and feeds into the LoRa chip directly as well.

I think if I remember right with the potential goal being that if a node is within range of 3 gateways that support GPS that then this could be used to then locate where the node is. But I haven't tried this out myself.

closed 2020-11-16 by ryanteck