PiSupply/IoTLoRaRange #21: Rx2 timeout Error

kranthisai, 2020-10-14:
Hi,
I get Rx2 timeout error. I followed the instructions from [here](https://learn.pi-supply.com/make/getting-started-with-the-raspberry-pi-lora-node-phat/)  for initial setup. Here is my lora_node.py 
`
#!/usr/bin/env python3
from rak811 import Mode, Rak811

lora = Rak811()
lora.hard_reset()
lora.mode = Mode.LoRaWan
lora.band = 'US915'
lora.set_config(ch_mask = '0,FF00')
lora.set_config(ch_mask = '1,0000')
lora.set_config(ch_mask = '2,0000')
lora.set_config(ch_mask = '3,0000')
lora.set_config(ch_mask = '4,0000')
lora.set_config(app_eui='70BXXXXXXXXX',app_key='EB37CXXXXXXXXXXXXXXXXXXXXX')
lora.join_otaa()
lora.dr = 5
lora.send('Hello world')
lora.close()
`
[image]

LoRa Gateway is up and can see online in things network.

ryanteck, 2020-10-14:
Hi @kranthisai , I believe we got a ticket which we were looking into. However we'll move it to here for quicker response.

Typically this issue is caused by the node not receiving a response from the Gateway, on the TTN console are you seeing any data on the Log? (Ideally if you could post a screenshot of the traffic that would be ideal).

Could you try moving the ch_mask lines underneath the line where you set your keys at the same time to see if that resolves the issue.

shawaj, 2020-10-14:
@ryanteck yep I asked @kranthisai to post it here instead so you could see it as well :-)

kranthisai, 2020-10-14:
Thanks @ryanteck @shawaj for quick response.  

I dont see any traffic in the Gateway Logs
[image]

I moved the lines ch_mask underneath keys, same output.

ryanteck, 2020-10-14:
Hmm ok, that's interesting.

Could you try changing the line with '''lora.dr''' from 5 to 3 and try that?

I think that could be why and seems to be someting we missed from the documentation.

kranthisai, 2020-10-14:
Actually I tried that before. I just tried again. same issue. Also tried to move lora.dr above join_otaa().

ryanteck, 2020-10-14:
Hmm, it should be 3 for US so I'd keep it at that for now as 5 won't work.

As for being above otaa I don't think that would work.

Is your gateway in the same room as the node? Is it setup for the correct region plan for TTN too?

Thanks

kranthisai, 2020-10-14:
Yes, Gateway is in same room.  Both are setup for us-west region.

ryanteck, 2020-10-14:
Ok so shouldn't be range issues then.

What brand is your gateway / have you been able to confirm if your gateway is succesfully receviing data from other nodes?

kranthisai, 2020-10-14:
We are using MokoSmart Gateway. Yes, We see the data from other sensors. 
Here is the traffic from other sensors.
[image]

ryanteck, 2020-10-14:
Ok that's handy to know. Looks like it's on the same plan. (As not all US gateways use the same channels).

I do see that it's using SF9 the other sensor so a DR of 1 might be worth trying.

For debugging I would also usually ask that you try the ABP example as well as sometimes it's easier to rectifiy the issue using ABP to then switch back to OTAA.

kranthisai, 2020-10-14:
tried DR 1 , same error. I will try with ABP

ryanteck, 2020-10-14:
Thanks, it's not a common error this and also being US915 is harder for us to debug being in Europe.

kranthisai, 2020-10-14:
I tried with ABP, I dont see any issue in joining to network, however I dont see any gateway traffic

[image]

[image]

ryanteck, 2020-10-18:
Hi @kranthisai ,

I will try to replicate this issue on Tuesday

kranthisai, 2020-10-22:
Hi @ryanteck , 

 Thanks. Let me know if you have any update.

ryanteck, 2020-10-22:
@kranthisai sorry I haven't had a chance to this week yet. I'll see if I can have a look at it over the weekend.

kranthisai, 2020-10-22:
Thanks @ryanteck

ryanteck, 2020-10-26:
@kranthisai Unfortunately I didn't get chance over the weekend, but I've started work on trying to replicate this issue this evening. I'll try and get back to you shortly with my results.

ryanteck, 2020-10-27:
Hi @kranthisai ,

I think I've successfully replicated the issue. I'll now work and see if I can resolve it.

kranthisai, 2020-10-27:
Awesome. Thanks

ryanteck, 2020-10-27:
Ok, so with OTAA I'm experiencing issues but with ABP I'm not.

With ABP, this following code works for me:

```
#!/usr/bin/env python3
from rak811 import Mode, Rak811

lora = Rak811()
lora.hard_reset()
lora.mode = Mode.LoRaWan
lora.band = 'US915'
lora.set_config(ch_mask = '0,FF00')
lora.set_config(ch_mask = '1,0000')
lora.set_config(ch_mask = '2,0000')
lora.set_config(ch_mask = '3,0000')
lora.set_config(ch_mask = '4,0000')
lora.set_config(dev_addr="", nwks_key="", apps_key="")

lora.join_abp()

lora.dr = 3

lora.send('Hello world')
lora.close()
```

I'll continue to try and see what I need to do to get OTAA working.

ryanteck, 2020-10-27:
Hi @kranthisai ,

It turns out actually OTAA is working fine on my unit too. I had incorrectly configured my gateway.

My OTAA code is 

```

#!/usr/bin/env python3
from rak811 import Mode, Rak811

lora = Rak811()
lora.hard_reset()
lora.mode = Mode.LoRaWan
lora.band = 'US915'
lora.set_config(ch_mask = '0,FF00')
lora.set_config(ch_mask = '1,0000')
lora.set_config(ch_mask = '2,0000')
lora.set_config(ch_mask = '3,0000')
lora.set_config(ch_mask = '4,0000')
lora.set_config(dev_eui='',app_eui='',app_key='')

#lora.dr = 3

lora.join_otaa()

lora.dr = 3

lora.send('Hello world')
lora.close()

```

My gateway log is as shown

[image]

ryanteck, 2020-11-03:
Hi @kranthisai ,

I was wondering if you had a chance to test the above code and let me know the output?

kranthisai, 2020-11-03:
Hi Ryan,
 I will test today and let you know.

Thanks

kranthisai, 2020-11-05:
Sorry. didn't get a chance. Will update over the weekend.

ryanteck, 2020-11-05:
No worries I'm the same sometimes!

kranthisai, 2020-11-09:
Not sure, what it was its not working with our gateway. We will buy things network lora gateway and test on that.

ryanteck, 2020-11-15:
Hi @kranthisai ,

I've only recently found out, for the Dev EUI how are you getting this?

I've been informed that on new batches there may be a sticker on the unit that says it, but our guide says to use the software's output and that these might mismatch.

panban-ux, 2020-11-16:
How is it possible to sent something parsed out of a JSON file. How is the correct syntax on having an uplink after a print variable?
On line 9 i have my print function were it prints the output of a JSON file narrowed to what i need. How will i convey this at lora.send without the error of line 20?
[image]

ryanteck, 2020-11-16:
Hi @panban-ux , this doesn't look like the same issue so I'll move it to it's own ticket.

kranthisai, 2020-11-17:
Hi @ryanteck ,

 I used the software output from the docs to get Dev EUI. I see there is a mismatch. I should use from the one on sticker?

Thanks,
Kranthi

ryanteck, 2020-11-17:
Hi @kranthisai ,

The software one would be the correct one, I just wanted to check as another customer had an issue but was using the one on the sticker not the software. 

That rules that out being an issue. Hmm

panban-ux, 2020-11-17:
i confirm on working using the sticker devEUI

ryanteck, 2020-11-17:
@panban-ux I believe this is because in your code you set the variable dev_eui with the one on the sticker which would work as it changes the dev_eui back to the one on the sticker.

As default they are different hence me checking.