PiSupply/IoTLoRaRange #22: Gateway not connected TTN

jfnunez-dev, 2020-10-21:
linaro@tinkerboard011:/opt/iotloragateway$ ./mp_pkt_fwd
18:28:33  *** Multi Protocol Packet Forwarder for Lora Gateway ***
Version: 3.0.20
18:28:33  *** Lora concentrator HAL library version info ***
Version: 5.0.1; Options: native;
***
18:28:33  INFO: Little endian host
18:28:33  INFO: found global configuration file /opt/iotloragateway/global_conf.json, parsing it
18:28:33  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
18:28:33  INFO: lorawan_public 1, clksrc 1
18:28:33  INFO: no configuration for LBT
18:28:33  INFO: antenna_gain 0 dBi
18:28:33  INFO: Configuring TX LUT with 16 indexes
18:28:33  INFO: radio 0 enabled (type SX1257), center frequency 917200000, RSSI offset -166.000000, tx enabled 1
18:28:33  INFO: radio 1 enabled (type SX1257), center frequency 917900000, RSSI offset -166.000000, tx enabled 0
18:28:33  INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
18:28:33  INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
18:28:33  INFO: FSK channel 8 disabled
18:28:33  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
18:28:33  INFO: Found 1 servers in array.
18:28:33  INFO: Server 0 configured to "thethings.meshed.com.au"
18:28:33  INFO: packets received with a valid CRC will be forwarded
18:28:33  INFO: packets received with a CRC error will NOT be forwarded
18:28:33  INFO: packets received with no CRC will NOT be forwarded
18:28:33  INFO: GPS is disabled
18:28:33  INFO: Upstream data is enabled
18:28:33  INFO: Downstream data is enabled
18:28:33  INFO: Ghoststream data is disabled
18:28:33  INFO: Radiostream data is enabled
18:28:33  INFO: Statusstream data is enabled
18:28:33  INFO: Beacon is disabled
18:28:33  INFO: Packet logger is disabled
18:28:33  INFO: Flush output after statistic is disabled
18:28:33  INFO: Flush after each line of output is disabled
18:28:33  INFO: Watchdog is disabled
18:28:33  INFO: found local configuration file /opt/iotloragateway/local_conf.json, parsing it
18:28:33  INFO: redefined parameters will overwrite global parameters
18:28:33  INFO: /opt/iotloragateway/local_conf.json does not contain a JSON object named SX1301_conf
18:28:33  INFO: /opt/iotloragateway/local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
18:28:33  INFO: gateway MAC address is configured to 0000000000000000
18:28:33  INFO: Found 1 servers in array.
18:28:33  INFO: Server 0 configured to "thethings.meshed.com.au"
18:28:33  INFO: packets received with a valid CRC will be forwarded
18:28:33  INFO: packets received with a CRC error will NOT be forwarded
18:28:33  INFO: packets received with no CRC will NOT be forwarded
18:28:33  INFO: GPS serial port path is configured to "/dev/serial1"
18:28:33  INFO: Reference latitude is configured to -33.044258 deg
18:28:33  INFO: Reference longitude is configured to -71.373367 deg
18:28:33  INFO: Reference altitude is configured to 10 meters
18:28:33  INFO: GPS is disabled
18:28:33  INFO: Upstream data is enabled
18:28:33  INFO: Downstream data is enabled
18:28:33  INFO: Ghoststream data is disabled
18:28:33  INFO: Radiostream data is enabled
18:28:33  INFO: Statusstream data is enabled
18:28:33  INFO: Beacon is disabled
18:28:33  INFO: Packet logger is disabled
18:28:33  INFO: Flush output after statistic is disabled
18:28:33  INFO: Flush after each line of output is disabled
18:28:33  INFO: Watchdog is disabled
18:28:33  INFO: Contact email configured to "jose.nunez@citysense.co"
18:28:33  INFO: Description configured to "Compile tutorial Teset"
18:28:33  INFO: [Transports] Initializing protocol for 1 servers
18:28:35  INFO: [TTN] server "thethings.meshed.com.au" connected
18:28:35  INFO: [main] Starting the concentrator
18:28:38  INFO: [main] concentrator started, radio packets can now be received.
18:28:38  INFO: JIT thread activated.
18:28:38  INFO: [up] Thread activated for all servers.
18:28:38  INFO: Disabling GPS mode for concentrator's counter...
18:28:38  INFO: host/sx1301 time offset=(1603315716s:303815µs) - drift=129739719µs
18:28:38  INFO: Enabling GPS mode for concentrator's counter.

##### 2020-10-21 21:29:08 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 0 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### BEACON IS DISABLED!
### [JIT] ###
# INFO: JIT queue contains 0 packets.
# INFO: JIT queue contains 0 beacons.
### GPS IS DISABLED!
### [PERFORMANCE] ###
# Upstream radio packet quality: 0.00%.
### [ CONNECTIONS ] ###
# thethings.meshed.com.au: Connected
# Semtech status report send.
##### END #####
18:29:08  INFO: [TTN] thethings.meshed.com.au RTT 460
18:29:09  INFO: [TTN] send status success for thethings.meshed.com.au
18:29:38  INFO: Disabling GPS mode for concentrator's counter...
18:29:38  INFO: host/sx1301 time offset=(1603315729s:342324µs) - drift=13038509µs
18:29:38  INFO: Enabling GPS mode for concentrator's counter.

##### 2020-10-21 21:29:39 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 0 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### BEACON IS DISABLED!
### [JIT] ###
# INFO: JIT queue contains 0 packets.
# INFO: JIT queue contains 0 beacons.
### GPS IS DISABLED!
### [PERFORMANCE] ###
# Upstream radio packet quality: 0.00%.
### [ CONNECTIONS ] ###
# thethings.meshed.com.au: Connected
# Semtech status report send.
##### END #####
18:29:39  INFO: [TTN] thethings.meshed.com.au RTT 622
18:29:40  INFO: [TTN] send status success for thethings.meshed.com.au

##### 2020-10-21 21:30:10 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 0 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### BEACON IS DISABLED!
### [JIT] ###
# INFO: JIT queue contains 0 packets.
# INFO: JIT queue contains 0 beacons.
### GPS IS DISABLED!
### [PERFORMANCE] ###
# Upstream radio packet quality: 0.00%.
### [ CONNECTIONS ] ###
# thethings.meshed.com.au: Connected
# Semtech status report send.
##### END #####
18:30:10  INFO: [TTN] thethings.meshed.com.au RTT 570
18:30:10  INFO: [TTN] send status success for thethings.meshed.com.au
18:30:38  INFO: Disabling GPS mode for concentrator's counter...
18:30:38  INFO: host/sx1301 time offset=(1603315729s:344400µs) - drift=2076µs
18:30:38  INFO: Enabling GPS mode for concentrator's counter.

##### 2020-10-21 21:30:40 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 0 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### BEACON IS DISABLED!
### [JIT] ###
# INFO: JIT queue contains 0 packets.
# INFO: JIT queue contains 0 beacons.
### GPS IS DISABLED!
### [PERFORMANCE] ###
# Upstream radio packet quality: 0.00%.
### [ CONNECTIONS ] ###
# thethings.meshed.com.au: Connected
# Semtech status report send.
##### END #####
18:30:40  INFO: [TTN] thethings.meshed.com.au RTT 551
18:30:42  INFO: [TTN] send status success for thethings.meshed.com.au

Any help?

jfnunez-dev, 2020-10-21:
linaro@tinkerboard011:~$ systemctl status iot-lora-gateway.service
● iot-lora-gateway.service - IOT LoRa Gateway Packet Forwarder
   Loaded: loaded (/lib/systemd/system/iot-lora-gateway.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-10-21 18:28:46 -03; 17min ago
  Process: 5978 ExecStartPre=/opt/iotloragateway/reset-22.sh (code=exited, status=0/SUCCESS)
 Main PID: 6032 (mp_pkt_fwd)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/iot-lora-gateway.service
           └─6032 /opt/iotloragateway/mp_pkt_fwd

Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]:  Upstream radio packet quality: 0.00%.
Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]:  [ CONNECTIONS ] ###
Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]:  thethings.meshed.com.au: Connected
Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]:  Semtech status report send.
Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]:  END #####
Oct 21 18:45:43 tinkerboard011 mp_pkt_fwd[6032]: 18:45:43  INFO: [TTN] thethings.meshed.com.au RTT 527
Oct 21 18:45:44 tinkerboard011 mp_pkt_fwd[6032]: 18:45:44  INFO: [TTN] send status success for thethings.meshed.com.au
Oct 21 18:45:51 tinkerboard011 mp_pkt_fwd[6032]: 18:45:51  INFO: Disabling GPS mode for concentrator's counter...
Oct 21 18:45:51 tinkerboard011 mp_pkt_fwd[6032]: 18:45:51  INFO: host/sx1301 time offset=(1603315729s:328884µs) - drift=826µs
Oct 21 18:45:51 tinkerboard011 mp_pkt_fwd[6032]: 18:45:51  INFO: Enabling GPS mode for concentrator's counter.

status service

ryanteck, 2020-10-21:
Could you provide full information as to what hardware you are using, what software and guides you're following to get this issue.

Thank you

jfnunez-dev, 2020-10-21:
i followed this instructions:
https://learn.pi-supply.com/make/how-to-compile-the-lora-gateway-software-for-the-lora-gateway-hat/

i am using a TinkerBoardS 
Linux tinkerboard011 4.4.132+ #8 SMP Thu Aug 13 08:51:08 UTC 2020 armv7l GNU/Linux

ryanteck, 2020-10-21:
We don't really support tinker board as it says in the top of those instructions.

However there doesn't actually look like there is any issues and I suspect that it could be something on TTNs side.

Could you try setting a different router in the config, possibly either the ttn-router-asia-se or ttn-router-jp routers as I suspect that it's an issue with using the meshed router.

jfnunez-dev, 2020-10-21:
i know nobody support tinkerboard, i am trying to test this hat, others hat no even make the gateway start but this do that, it's weird that appear to connected in the log but in TTN don't.
i dont know if change the local_conf.json to legacy mode helps, maybe its something with gateway ID or gateway EUI

ryanteck, 2020-10-21:
If you could try one of the 2 other routers (Or even TTN's EU one) as a try as I do remember some issue with the meshed router working differently.

The other test would be are you just looking to see if the gateway is showing as online in the console or have you also got a node transmitting packets. As for a while TTN has had intermittent issues with the online status not being 100% accurate (https://status.thethings.network/)

ryanteck, 2020-10-21:
Actually, for the meshed router you might need to use the console at https://console.thethings.meshed.com.au/ @jfnunez-dev to see the status. If you could also check there.

jfnunez-dev, 2020-10-21:
i changed the console and to the US router and keep to not connected 

linaro@tinkerboard011:/opt/iotloragateway$ ./mp_pkt_fwd
19:48:33  *** Multi Protocol Packet Forwarder for Lora Gateway ***
Version: 3.0.20
19:48:33  *** Lora concentrator HAL library version info ***
Version: 5.0.1; Options: native;
***
19:48:33  INFO: Little endian host
19:48:33  INFO: found global configuration file /opt/iotloragateway/global_conf.json, parsing it
19:48:33  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
19:48:33  INFO: lorawan_public 1, clksrc 1
19:48:33  INFO: no configuration for LBT
19:48:33  INFO: antenna_gain 0 dBi
19:48:33  INFO: Configuring TX LUT with 16 indexes
19:48:33  INFO: radio 0 enabled (type SX1257), center frequency 904300000, RSSI offset -166.000000, tx enabled 1
19:48:33  INFO: radio 1 enabled (type SX1257), center frequency 905000000, RSSI offset -166.000000, tx enabled 0
19:48:33  INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
19:48:33  INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
19:48:33  INFO: FSK channel 8 disabled
19:48:33  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
19:48:33  INFO: Found 1 servers in array.
19:48:33  INFO: Server 0 configured to "router.us.thethings.network"
19:48:33  INFO: packets received with a valid CRC will be forwarded
19:48:33  INFO: packets received with a CRC error will NOT be forwarded
19:48:33  INFO: packets received with no CRC will NOT be forwarded
19:48:33  INFO: GPS is disabled
19:48:33  INFO: Upstream data is enabled
19:48:33  INFO: Downstream data is enabled
19:48:33  INFO: Ghoststream data is disabled
19:48:33  INFO: Radiostream data is enabled
19:48:33  INFO: Statusstream data is enabled
19:48:33  INFO: Beacon is disabled
19:48:33  INFO: Packet logger is disabled
19:48:33  INFO: Flush output after statistic is disabled
19:48:33  INFO: Flush after each line of output is disabled
19:48:33  INFO: Watchdog is disabled
19:48:33  INFO: found local configuration file /opt/iotloragateway/local_conf.json, parsing it
19:48:33  INFO: redefined parameters will overwrite global parameters
19:48:33  INFO: /opt/iotloragateway/local_conf.json does not contain a JSON object named SX1301_conf
19:48:33  INFO: /opt/iotloragateway/local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
19:48:33  INFO: gateway MAC address is configured to 0000000000000000
19:48:33  INFO: Found 1 servers in array.
19:48:33  INFO: Server 0 configured to "router.us.thethings.network"
19:48:33  INFO: packets received with a valid CRC will be forwarded
19:48:33  INFO: packets received with a CRC error will NOT be forwarded
19:48:33  INFO: packets received with no CRC will NOT be forwarded
19:48:33  INFO: GPS serial port path is configured to "/dev/serial1"
19:48:33  INFO: Reference latitude is configured to -33.044258 deg
19:48:33  INFO: Reference longitude is configured to -71.373367 deg
19:48:33  INFO: Reference altitude is configured to 10 meters
19:48:33  INFO: GPS is disabled
19:48:33  INFO: Upstream data is enabled
19:48:33  INFO: Downstream data is enabled
19:48:33  INFO: Ghoststream data is disabled
19:48:33  INFO: Radiostream data is enabled
19:48:33  INFO: Statusstream data is enabled
19:48:33  INFO: Beacon is disabled
19:48:33  INFO: Packet logger is disabled
19:48:33  INFO: Flush output after statistic is disabled
19:48:33  INFO: Flush after each line of output is disabled
19:48:33  INFO: Watchdog is disabled
19:48:33  INFO: Contact email configured to "jose.nunez@citysense.co"
19:48:33  INFO: Description configured to "Compile tutorial Teset"
19:48:33  INFO: [Transports] Initializing protocol for 1 servers
19:48:36  INFO: [TTN] server "router.us.thethings.network" connected
19:48:36  INFO: [main] Starting the concentrator
19:48:39  INFO: [main] concentrator started, radio packets can now be received.
19:48:39  INFO: [up] Thread activated for all servers.
19:48:39  INFO: JIT thread activated.
19:48:39  INFO: Disabling GPS mode for concentrator's counter...
19:48:39  INFO: host/sx1301 time offset=(1603320517s:449784µs) - drift=635918392µs
19:48:39  INFO: Enabling GPS mode for concentrator's counter.

 2020-10-21 22:49:09 GMT #####
 [UPSTREAM] ###
 RF packets received by concentrator: 1
 CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
 RF packets forwarded: 0 (0 bytes)
 PUSH_DATA datagrams sent: 0 (0 bytes)
 PUSH_DATA acknowledged: 0.00%
 [DOWNSTREAM] ###
 PULL_DATA sent: 0 (0.00% acknowledged)
 PULL_RESP(onse) datagrams received: 0 (0 bytes)
 RF packets sent to concentrator: 0 (0 bytes)
 TX errors: 0
 BEACON IS DISABLED!
 [JIT] ###
 INFO: JIT queue contains 0 packets.
 INFO: JIT queue contains 0 beacons.
 GPS IS DISABLED!
 [PERFORMANCE] ###
 Upstream radio packet quality: 0.00%.
 [ CONNECTIONS ] ###
 router.us.thethings.network: Connected
 Semtech status report send.
 END #####
19:49:09  INFO: [TTN] router.us.thethings.network RTT 261
19:49:09  ERROR: [TTN] send status failed for router.us.thethings.network
19:49:09  INFO: [TTN] Reconnecting router.us.thethings.network
19:49:11  INFO: [TTN] server "router.us.thethings.network" connected

and everything look good

ryanteck, 2020-10-21:
The issue with the US one is that the domain isn't correct as it isn't just US, if you wish to use US it would be I believe "router.us-west.thethings.network" and changing it to that in the TTN Console too.

Another you could try is "router.eu.thethings.network" for the EU one, as currently I know it's working fine as that's what I'm using here in the UK.

jfnunez-dev, 2020-10-21:
if change the router the output its different

linaro@tinkerboard011:/opt/iotloragateway$ ./mp_pkt_fwd
20:00:05  *** Multi Protocol Packet Forwarder for Lora Gateway ***
Version: 3.0.20
20:00:05  *** Lora concentrator HAL library version info ***
Version: 5.0.1; Options: native;
***
20:00:05  INFO: Little endian host
20:00:05  INFO: found global configuration file /opt/iotloragateway/global_conf.json, parsing it
20:00:05  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
20:00:05  INFO: lorawan_public 1, clksrc 1
20:00:05  INFO: no configuration for LBT
20:00:05  INFO: antenna_gain 0 dBi
20:00:05  INFO: Configuring TX LUT with 16 indexes
20:00:05  INFO: radio 0 enabled (type SX1257), center frequency 904300000, RSSI offset -166.000000, tx enabled 1
20:00:05  INFO: radio 1 enabled (type SX1257), center frequency 905000000, RSSI offset -166.000000, tx enabled 0
20:00:05  INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
20:00:05  INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
20:00:05  INFO: FSK channel 8 disabled
20:00:05  INFO: /opt/iotloragateway/global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
20:00:05  INFO: Found 1 servers in array.
20:00:05  INFO: Server 0 configured to "router.us-west.thethings.network"
20:00:05  INFO: packets received with a valid CRC will be forwarded
20:00:05  INFO: packets received with a CRC error will NOT be forwarded
20:00:05  INFO: packets received with no CRC will NOT be forwarded
20:00:05  INFO: GPS is disabled
20:00:05  INFO: Upstream data is enabled
20:00:05  INFO: Downstream data is enabled
20:00:05  INFO: Ghoststream data is disabled
20:00:05  INFO: Radiostream data is enabled
20:00:05  INFO: Statusstream data is enabled
20:00:05  INFO: Beacon is disabled
20:00:05  INFO: Packet logger is disabled
20:00:05  INFO: Flush output after statistic is disabled
20:00:05  INFO: Flush after each line of output is disabled
20:00:05  INFO: Watchdog is disabled
20:00:05  INFO: found local configuration file /opt/iotloragateway/local_conf.json, parsing it
20:00:05  INFO: redefined parameters will overwrite global parameters
20:00:05  INFO: /opt/iotloragateway/local_conf.json does not contain a JSON object named SX1301_conf
20:00:05  INFO: /opt/iotloragateway/local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
20:00:05  INFO: gateway MAC address is configured to 0000000000000000
20:00:05  INFO: Found 1 servers in array.
20:00:05  INFO: Server 0 configured to "router.us-west.thethings.network"
20:00:05  INFO: packets received with a valid CRC will be forwarded
20:00:05  INFO: packets received with a CRC error will NOT be forwarded
20:00:05  INFO: packets received with no CRC will NOT be forwarded
20:00:05  INFO: GPS serial port path is configured to "/dev/serial1"
20:00:05  INFO: Reference latitude is configured to -33.044258 deg
20:00:05  INFO: Reference longitude is configured to -71.373367 deg
20:00:05  INFO: Reference altitude is configured to 10 meters
20:00:05  INFO: GPS is disabled
20:00:05  INFO: Upstream data is enabled
20:00:05  INFO: Downstream data is enabled
20:00:05  INFO: Ghoststream data is disabled
20:00:05  INFO: Radiostream data is enabled
20:00:05  INFO: Statusstream data is enabled
20:00:05  INFO: Beacon is disabled
20:00:05  INFO: Packet logger is disabled
20:00:05  INFO: Flush output after statistic is disabled
20:00:05  INFO: Flush after each line of output is disabled
20:00:05  INFO: Watchdog is disabled
20:00:05  INFO: Contact email configured to "jose.nunez@citysense.co"
20:00:05  INFO: Description configured to "Compile tutorial Teset"
20:00:05  INFO: [Transports] Initializing protocol for 1 servers
20:00:08  ERROR: [TTN] Connection to server "router.us-west.thethings.network" failed, retry in 30 seconds
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 0, total 30
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 10, total 30
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 20, total 30
20:00:39  ERROR: [TTN] Connection to server "router.us-west.thethings.network" failed, retry in 60 seconds
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 0, total 60
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 10, total 60

ryanteck, 2020-10-21:
Ok, that confirms there's something wrong with the US router in your case.

Could you try the EU one like I've suggested?

jfnunez-dev, 2020-10-21:
its working!! with the US router

jfnunez-dev, 2020-10-21:
but i need AU915 frequency plan for my country (Chile) :c

ryanteck, 2020-10-21:
Ahh excellent,

I think with the Austrialian Meshed router it might work but you have to then remember to use the meshed router's console instead.

For using TTNs console with full support then for your location you might find the Asia one slightly quicker but it varies on a lot of factors.

ryanteck, 2020-10-21:
The router used doesn't affect the frequency plan, as long as you downloaded the AU915 plan when setting up the gateway, and have AU915 on the console then it should work fine.

ryanteck, 2020-10-21:
Also being in Chile you might find the brazil router the fastest, (router.brazil.thethings.network). The frequency plan of AU915 will still work as explained above.

jfnunez-dev, 2020-10-21:
thanks for the help, i gonna keep working on that, i need to getting working this device for Chile

ryanteck, 2020-10-21:
As I said it should work perfectly fine with AU915 no matter which router is selected, as long as you downloaded the AU global_conf and have it set to Australia frequency plan in the console still.

Can I mark this issue as resolved as it's working now?

jfnunez-dev, 2020-10-21:
yes, thank you very much

closed 2020-10-21 by ryanteck

ryanteck, 2020-10-27:
Hi @jfnunez-dev ,

I'm sorry to say that I did make a mistake with the router selection.

While the downlink (Node to gateway) configures based on which frequency plan is selected, it seems the Uplink (Gateway to Node) is configured based on server.

I didn't realise this until now so I apologise if this has caused you any issues.