PiSupply/IoTLoRaRange #23: gateway connected and disconnected after a few minutes

jfnunez-dev, 2020-10-26:
[image]

i getting this problem where the gateways connected and disconnected, i am using a TinkerBoardS with the hat and a Enterprise Account on TTN

ryanteck, 2020-10-26:
Do you have any information from the error logs on the gateway itself? I can't realy help out on what you've provided.

As previously stated Tinkerboard + Not using our software limits our support options.

I would guess however that your internet connection to TTN is dropping out causing these issues. You say enterprise TTN, do you mean The Things Industries?

jfnunez-dev, 2020-10-26:
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: ### [PERFORMANCE] ###
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: # Upstream radio packet quality: 100.00%.
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: ### [ CONNECTIONS ] ###
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: # citysense.eu1.cloud.thethings.industries: Connected
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: # Semtech status report send.
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: ##### END #####
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: 13:40:45  INFO: [TTN] citysense.eu1.cloud.thethings.industries RTT 310
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: 13:40:45  ERROR: [TTN] send status failed for citysense.eu1.cloud.thethings.industries
Oct 26 13:40:45 tinker_rak mp_pkt_fwd[4634]: 13:40:45  INFO: [TTN] Reconnecting citysense.eu1.cloud.thethings.industries
Oct 26 13:40:51 tinker_rak mp_pkt_fwd[4634]: 13:40:51  ERROR: [TTN] Connection to server "citysense.eu1.cloud.thethings.industries" failed,

Yes, The Things Industries 
i want to discart any "time dead" or time sleep of the script mp_pkt_fwd or its something else, i have no other problem with the TinkerBoardS only this connected/disconnected issue.
if its something with the server i can investigate with TTI, only need to discard hardware/software issue with the hat.

jfnunez-dev, 2020-10-26:
i have others gateways and i don't see this issue (Rak,Kerlink,Dragino)

ryanteck, 2020-10-26:
It says in the error log it's failing to communicate with TTI. So that is likely the cause. 

The hardware itself is working fine as indicated by the fact the software itself isn't crashing.

jfnunez-dev, 2020-10-26:
maybe its a latency issue with TTI, if there isnt any problem or crash with the script the hardware/software is discarded in the issue, thank for the help, greetings!

closed 2020-10-26 by jfnunez-dev