PiSupply/IoTLoRaRange #24: Gateway EUI?

jfnunez-dev, 2020-11-02:
if a need to register the gateway with gateway eui parameter in TTN , how i need to do it? i am only see register by gateway ID in the tutorial, it is necessary to modify the local_conf.json file with adding or modify the parameter ?

ryanteck, 2020-11-02:
Hi @jfnunez-dev ,

From what I remember you're using the compilie it yourself method, and yes it is possible to configure it in "legacy" mode.

To do this you'd need to do the following:

Change serv_type to "semtech"

Remove the lines "serv_gw_id" and "serv_gw_key".

Create your own EUI and change the line "gateway_ID" to the EUI you create and then set this up on the LoRa provider.

Typically we suggest using the mac address of the gateway, so if my mac address is "7c:dd:90:4f:fa:61" you'd first remove the :s making it "7cdd904ffa61" and then in the middle (6 characters in) adding FFFF resulting in my EUI being "7CDD90FFFF4FFA61".

jfnunez-dev, 2020-11-02:
how i get the mac address of the gateway from this product?

ryanteck, 2020-11-02:
Sorry, I misworded. By gateway I meant the host you're using (So the Tinkerboard) and using the mac address of the ethernet controller. (Typically found by using ifconfig).

ryanteck, 2020-11-03:
Hi @jfnunez-dev ,

Did this resolve your issue?

jfnunez-dev, 2020-11-06:
hi, i dont make any test yet with EUI, i am verifying with TTI for futures gateways hats

Best regards

closed 2020-11-06 by jfnunez-dev