PiSupply/IoTLoRaRange #26: How is it possible to sent something parsed out of a JSON file. How is the correct syntax on having an uplink after a print variable?

ryanteck, 2020-11-16:
How is it possible to sent something parsed out of a JSON file. How is the correct syntax on having an uplink after a print variable?
On line 9 i have my print function were it prints the output of a JSON file narrowed to what i need. How will i convey this at lora.send without the error of line 20?
[image]

_Originally posted by @panban-ux in https://github.com/PiSupply/IoTLoRaRange/issues/21#issuecomment-728038899_

ryanteck, 2020-11-16:
Hi @panban-ux
The send command takes byte arrays or strings with strings being the easiest.

You're trying to send a json array, so you'd need to encode it into a more suitable format. Potentially a custom byte array would work such as assigning each key a hex value and then formatting it as keyHex, Value for each one. 

I hope this helps.

panban-ux, 2020-11-16:
Ryanteck i used after this example an:
a=str(counter.most_common())
and then
lora.sent(a)
but still have an error.
How will i Hexify it? The above is a string retain, isnt it?
Thanks for moving the subject

ryanteck, 2020-11-16:
Hi @panban-ux ,

From the image where you print the counter.most_common() part above the LoRa code shows that it's printing if I'm correct a python List with tuples inside of it. You should be able to confirm it if you run `type(counter.most_common())` should output the name of the data.

For this type of data I would say you'd need to put together some kind of byte array or custom string as just converting the list to a string would likely be too many characters for LoRa.

panban-ux, 2020-11-16:
I didnt got the type(counter.most_common()). 
Anyway either way if i sent an "a" character i get the 61 in the TTN. Hence it encodes it that way. Am i wrong?

ryanteck, 2020-11-16:
That's correct but that's because the a is a string, the value you're trying to send is a python list so you'd need to first convert to a string.

However the resulting string may then be too long so it might not work. But is worth a shot.

closed 2020-12-23 by ryanteck