PiSupply/IoTLoRaRange #29: Network detection failed with RAK2247

tank01, 2020-12-21:
Hi team,

I followed the steps described here to install on RAK2247 :
https://learn.pi-supply.com/make/iot-lora-gateway-sd-card-image-setup/
I'm blocked in step 5, cant reach http://iotloragateway.local .

The Red light is on and the Green LED Blink on the Rpi board. The network green and orange led stays on. On my router, the port connected to the Rpi is on during few minutes, then it is off.

Could it be a problem with the Rpi ethernet port? Or because the SD Card image is only for RAK833, not RAK2247?

BR

shawaj, 2020-12-21:
@tank01 Do you have the Pi plugged in to a display and a keyboard?

If so, on the terminal could you run `ifconfig` and then see what IP address you have for eth0 interface. You should also be able to access the dashboard from this IP address.

Just to check, you are trying to access the iotloragateway.local access from a different computer right? Which operating system does this other computer use?

ryanteck, 2020-12-21:
Hi @tank01 as @shawaj said one diagnostics step would be to try getting the IP off the pi itself if you plug a keyboard and screen in. 

The 2247 works identically to the 833 so that's not causing the issue and some other part is acting up.

The only thng I can think to ask is what model of Raspberry Pi are you using?

shawaj, 2020-12-21:
@ryanteck FYI if accessing from windows, iotloragateway.local won't work out of the box unless you have skype installed or have separately installed bonjour. Because it can't handle mDNS very well.

ryanteck, 2020-12-21:
Yeah the document has to use IP address if on windows.

V2 I've got this fixed by using a different method.

tank01, 2020-12-21:
I'm using a computer on Ubuntu.
I've ordered a minihdmi screen cable to plug my Rpi. As soon as I received it, I'll come back to you.
Thank!

ryanteck, 2020-12-21:
Hi @tank01 ,

That should be working with Ubuntu then no issue.

Could you confirm which model of Raspberry Pi you are using?

shawaj, 2020-12-21:
FWIW I've had issues with .local addresses on Android previously. Not sure why.

tank01, 2020-12-23:
Hi all,
I received my screen cable :-)

It seems to be an issue with the SD Card image, check bellow 
[image]

Any idea?

ryanteck, 2020-12-23:
@tank01 ,

Could you confirm what model of Raspberry Pi you are using?

tank01, 2020-12-23:
Raspberry Pi 4 B 2GB

ryanteck, 2020-12-23:
Oh interesting, I've only seen this issue with an 8GB Raspberry Pi before.

Could you take a picture of the Raspberry Pi without the HAT on for me? I have a feeling that something has changed.

tank01, 2020-12-23:
One of nylon screw is broken :-(
Do you have the info on the bellow part ?
[image]

tank01, 2020-12-23:
Is it visble here ?
[image]

ryanteck, 2020-12-23:
Hi @tank01 ,

That gives me enough of a view to tell. It is a slightly later revision however Its the first I've seen that the SD card image won't work with them.

My next suggestion would be to re-burn the SD Card if you haven't already.  Just to check that you used the 2019-07-31-1 version?

If not it would be that I need to get a new version out. Currently the older version only hasn't worked on 8GB models so it's not been at the top of my list. However not working with 2GB models now is a more significant issue so I'll need to work on that sooner.

It's most likely going to be early to mid January before I can get this out.

tank01, 2020-12-23:
Thank you for your precise reply @ryanteck .

I confirm the SD card was burned (2 times) with 2019-07-31-iot-lora-gateway-lite.img (latest release), but the same problem appears.

Please let me know when the new version is released. Hope to hear from you soon, good luck!

tank01, 2021-01-12:
Hi @ryanteck , hope you are weel!
I need to go ahead with my projet.
Could you tell me if it will be ready as planned?

ryanteck, 2021-01-12:
Hi @tank01 ,

I'm sorry but I don't have a new ETA as my workload has been shifted to another project.

Do you have an older Raspberry Pi at all?

tank01, 2021-01-12:
No don't have older available PRI . Do you have suggestions? Any idea where I can find the right herware?

ryanteck, 2021-01-12:
I'll try and see if I can come up with a workaround tomorrow and get back to you.

tank01, 2021-01-13:
Ok, thanks, let me know

tank01, 2021-01-18:
Hi @ryanteck , any update? I need to finish this project. Do you know where I can buy the right hardware?

shawaj, 2021-01-18:
@tank01 do you have an SD card reader?

You should just be able to download the contents is the Raspberry Pi firmware repository (specifically start4.elf) and overwrite what is currently on the card

ryanteck, 2021-01-18:
Yeah that should fix it as @shawaj said.

I'll try first thing tomorrow to quickly make a new image with these files added and give it a test for you @tank01

tank01, 2021-01-19:
Ok perfect, let me know @ryanteck

ryanteck, 2021-01-19:
Unfortunately the quick fix that me and @shawaj thought should work currently with initial testing does not. 

I'll try to see what I can do as a workaround.

shawaj, 2021-01-19:
@ryanteck even with an updated kernel.img?

tank01, 2021-02-02:
Hi @ryanteck @shawaj 
Could you provide a solution in the coming 2 weeks?
If not possible in the code, could you tell me where I could buy the "legacy" hadware?

ryanteck, 2021-02-02:
Hi @tank01 ,

Unfortunately this is taking me longer to fix as we're also very busy with other products as well.

Using an older Raspberry Pi (Raspberry Pi 2, 3, or 3+) should work fine.