PiSupply/IoTLoRaRange #30: Support for The Things Network v3 clusters

AndrewJSchofield, 2021-01-31:
I've been using the Raspberry Pi IoT Gateway HAT with LoRa successfully with The Things Network. I've recently noticed a banner on the TTN web page that says my existing v2 cluster will be shut down in the future and I should migrate to v3. I don't know when that will happen but seemed sensible to investigate.

I've spent a few hours trying to get the configuration for the packet forwarder to work with v3. Once I'd created a v3 gateway on TTN, there was a button to download a global_conf.json file. This looked very similar to the supplied global_conf.json file already on my Pi, with the server information changed. Replacing the global_conf.json file didn't work directly because it's regenerated automatically so my changes were overwritten. A bit of fiddling stopped that from happening, but still the new v3 global_conf.json that I downloaded from TTN didn't do the trick.

Will there be an update to the IoT LoRa gateway SD card image to make it work with TTN v3? With better documentation, I think there's a good chance I could make it work, but I'm confused by the difference between a gateway ID, a gateway EUI, whether I need a TTN v3 gateway API key, and how I would supply it in the config.

Any help would be gratefully received.

ryanteck, 2021-01-31:
Hi Andrew,

This is something I'm going to try and look into this week, unfortunately I wasn't given any advance notice this change was going to happen and only found out on Thursday.

I believe that as an initial stop gap measure with just new documentation the current image should work fine with settings changes. I'm going to try and give this a go very shortly as I have a gateway setup on TTN myself that I need to migrate.

Currently however I have seen mixed reports as to when to start migrating gateways, while TTN say ASAP some of the community members are recommending leaving it slightly.

From what I've read Nodes which are upgraded to V3 still work with V2 gateways so these are recommended to be first transitioned which should be an easier process I believe, I'll also try to get our guides upgraded soon.

Then at a slightly later date the Gateway migration is better, as once a gateway is migrated to V3 I believe V2 sensors won't be processed by it.

AndrewJSchofield, 2021-02-28:
Hi, I have successfully configured my Arduino nano using the LoRa Shield with v3 and that works fine. I'll keep an eye on this issue for news about the gateway. Thanks, Andrew

HaccasDave, 2021-04-06:
Just wondering if there is any more news on this. I want to add a second gateway and a handful more nodes, but don't want to set it all up on TTN v2 only to have to redo it all.

Andrew - Is there anything specific that you did to move over to TTN v3?

Thanks.

bucklevision, 2021-04-11:
I'm keen on this too. On top of all of that though I'd love it if you could update the image to being based on Raspian 64 bit, as my pi is not only used as a gateway, but also needs other kit to run too and they all require a 64bit OS.

TheNetStriker, 2021-05-05:
I've just managed to connect my gateway to ttn v3. I did the following steps:

- Switch the `Packet Forwarder Config` to the `Semtech legacy mode`
- Enter the auto generated Gateway ID into the `Gateway EUI `field in the ttn v3 console and save the gateway
- Set description to anything
- Create an API token in ttn v3 console with the right **Link as Gateway to a Gateway Server for traffic exchange, i.e. write uplink and read downlink**
- Enter the API token into the `Gateway Key`
- Type 3 random characters after the `Gateway Key` (Else the UI will not accept this size of key)
- Enter region, location and save the configuration
- Edit the file **/opt/iotloragateway/local_conf.json** using  an ssh shell and remove the additional characters entered before.
- Restart the packet forwarder in the UI

After all those steps my gateway connected successfully to the ttn v3 network.

ryanbateman, 2021-05-10:
I had tried the same as @TheNetStriker with much the same result and am also keen in understanding whether an update is feasible. (Though I have also seen the same wariness and vagueness regarding the updating of gateways, so am not rushed at all.)

jakobfriis, 2021-06-02:
I built the software manually, and i cannot get it to connect. The built gateway works fine on TTN v2, but changing the config json files to point to TTN v3, with the API key and new server address does not work

16:37:18  INFO: Downstream data is enabled
16:37:18  INFO: Ghoststream data is disabled
16:37:18  INFO: Radiostream data is enabled
16:37:18  INFO: Statusstream data is enabled
16:37:18  INFO: Beacon is disabled
16:37:18  INFO: Packet logger is disabled
16:37:18  INFO: Flush output after statistic is disabled
16:37:18  INFO: Flush after each line of output is disabled
16:37:18  INFO: Watchdog is disabled
16:37:18  INFO: Contact email configured to "jf@jakobf.dk"
16:37:18  INFO: Description configured to "test"
16:37:18  INFO: [Transports] Initializing protocol for 1 servers
16:37:18  ERROR: [TTN] Connection to server "eu1.cloud.thethings.network" failed, retry in 30 seconds
src/ttn_transport.c:371:ttn_connect(): ttn_connect: sleeping() at 0, total 30

{
    "gateway_conf": {
        "gateway_ID": "tinesvej",
        "contact_email" : "jf@jakobf.dk",
        "description": "test",
         "servers": [
            {
                "serv_type": "ttn",
                "server_address": "eu1.cloud.thethings.network",
                "serv_gw_id": "3245531135677363",
                "serv_gw_key": "XXXXXXXXXXXXXXXXXXXX2UEWWWHI3WM2HADII6YWV3S2Q.IIHBS5NUMEE7S3YLWZA7ESQ26O7ZW3YRFHS6RGKJLUASKCEMKZVA",
                "serv_enabled": true
            }
        ],
        "gps": false,
  "fake_gps": true,
  "ref_latitude": 0,
  "ref_longitude": 0,
  "ref_altitude": 0,
  "gps_tty_path":"/dev/serial1"
    }
}

jakobfriis, 2021-06-02:
I have it connected now. I downloaded the global_conf.json from TTN v3, and added the serv_gw_key with the generated key.

ryanbateman, 2021-09-22:
Any official update on this?

ptbw, 2021-09-25:
Even some guidance on how to manually configure the global_config.json would be useful. I think I followed @jakobfriis method and I have a connection in the std out on the pi  but not seeing that status change in the v3 console.

(@jakobfriis could you post your global_config.json without your api key pls.)

Thanks

michal-lusiak, 2021-09-27:
@ryanteck Can we please get any update on the migration to The Things Stack Community Edition (v3)?

ptbw, 2021-09-27:
It may help some but I have switched to using this method for updating my Gateway to V3. 
https://github.com/kersing/ttn-resin-gateway-rpi-1

It seems to be working fine with the new ThingsNetwork community edition.

ryanbateman, 2021-12-06:
@ptbw Did you need to do any additional configuration/adjustment or did you just follow the instructions on the repo as-is, with Balena Cloud set and all?