PiSupply/IoTLoRaRange #32: RPi4 reducing core_freq

JeroenIoT, 2021-06-06:
To build a lora gw I use the LoRa hat from pi-supply (2247 SPI based)

In the pi-supply manuals it is written that the core_freq of the RPi4 must be reduced (change **core_freq to 250** in the config.txt).
However according to raspberry this is n**o longer supported** in the **RPi4** (see link below)

https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

Quote from raspberry pi org:
`Changing core_freq in config.txt is not supported on the Pi 4, any change from the default will almost certainly cause a failure to boot.`

Sp how to get the LoRa hat from pi-supply working?

tvoverbeek, 2021-06-06:
Remember the page talks about overclocking. The warning is for increasing the core frequency.
Table further down shows default core_freq = s500 and core_freq_min - 250 for Pi4.
So underclocking core_freq to 250 is no problem (have done it several times on a Pi4 to get a constant i2c clock)

JeroenIoT, 2021-06-07:
Thanks for your reply. 
I have not done it before but how would it work as (on the page I refer to above) it is explicitly stated that the core_freq changing in config.txt does not work

An example would be really appreciated

tvoverbeek, 2021-06-07:
The SPI clock is derived from the core (VideoCore) clock. When the core clock is varying depending on the load the SPI clock is also varying. You want a constant core clock independent of the load for the LoRa hat (= constant SPI clock).

The Pi4 warning on the overclocking page is misleading. Any of the min and default values are OK. Others are not.

Here is a simple script to show the core clock as function of the load:
``` bash
#! /bin/bash
sleep 1 && vcgencmd measure_clock core
while true; do true; done &
sleep 1 && vcgencmd measure_clock core
kill %1
```
On a default Pi4 (no enable_uart=1 and bluetooth not disabled in config.txt) thie outputs:
```
pi@raspberrypi:~ $ ./clocktest.sh 
frequency(1)=200008304
frequency(1)=499987808
pi@raspberrypi:~ $
```
core_freq changes from 200MHz to 500MHz when loaded.

In /boot/config.txt with
```
core_freq=250
core_freq_min=250
```
the script output is:
```
pi@raspberrypi:~ $ ./clocktest.sh 
frequency(1)=250000496
frequency(1)=250000496
pi@raspberrypi:~ $
```
This is the setting you want for the LoRa hat.

See also the discussion in https://github.com/raspberrypi/firmware/issues/1308

closed 2023-02-19 by JeroenIoT