PiSupply/IoTLoRaRange #36: The Things Network v3

hardillb, 2021-11-12:
Hi,

Are there any instructions for setting up the Gateway Hat with the new v3 Things network setup?

ryanbateman, 2022-03-02:
@hardillb FWIW, I managed to get it working using the [RAKWireless repo here](https://github.com/RAKWireless/rak_common_for_gateway), following the instructions there regarding standard setup, then downloading the `global_conf.json` from the v3 console and using that to update the packet forward config.

hardillb, 2022-03-02:
@ryanbateman Thanks, I have mine working via that route, but it would be nice for PiSupply offer an update since they are still selling this hardware.

ryanbateman, 2022-03-23:
@hardillb Yes, I very much agree.

salmanfarisvp, 2022-08-11:
Thnaks, @ryanbateman Could you please add the instructions and do a pull request if possible? Thanks.

ChristopherRush, 2022-08-11:
I have instructions for this already.

emendir, 2024-07-30:
It took me a bit of research and trial and error, but I succeeded in the end.
Over the last three years the configuration options on TheThingsNetwork have changed often, so I don't know how long they'll stay valid for.

Here are my notes in which I documented what I learned. They're not step-by-step instructions, but for people already familiar with working with LoRaWAN and TheThingsNetwork, they should give you all the info you need to get it working.

_LoRaWAN settings for working with [[LoRa Node pHAT RAK811]]_
#lora
## RAK811

The [[LoRa Node pHAT RAK811]] hardware and the [RAK811 python library](https://github.com/AmedeeBulle/pyrak811/tree/main)'s LoRaWAN compatibility seem to be limited to a specific set of its specifications.
I deduce this from the configuration options documented [here](https://github.com/AmedeeBulle/pyrak811/blob/main/DOCS/RAK811_AT_Command_Manual_V1.0.pdf) being a subset of all LoRaWAN specs.
Those configurations which I could not configure on the RAK811, I experimented with when trying to register the device on TheThinsNetwork, to find which configurations there resulting in the RAK811 working correctly.

### Supported Configurations

Here are the configurations I know I can apply:
- LoRaP2p
- LoRaWAN
	- [End Device Activation (OTAA, ABP)](https://www.thethingsnetwork.org/docs/lorawan/end-device-activation/)
	- [Device Class (A,C (class B is not supported))](https://www.thethingsnetwork.org/docs/lorawan/classes/)

Given that we're working with 
For these settings, I configured, on both the RAK811 and in the TheThingsNetwork console when registering a new device, the following:

| RAK Config Name | TTN Config Name | Value                             |
| --------------- | --------------- | --------------------------------- |
| join_mode       | Activation mode | OTAA                              |
| app_eui         | JoinEUI         | _I chose a value myself_          |
| app_key         | AppKey          | _generated by TTN_                |
| app_eui         | JoinEUI         | _from RAK811 or generated by TTN_ |

### Fixed TTN Configurations
Here are the configurations that can't be applied/I didn't apply to the RAK811 which I had to apply in TheThingsNetwork console (while adding a device):

| Configuration               | Value                                      |
| --------------------------- | ------------------------------------------ |
| Frequency Plan              | Europe 863-870 MHz (SF9 for RX2)           |
| LoRaWAN Version             | LoRaWAN Specification 1.0.3                |
| Regional Parameters version | RP001 Regional Parameters 1.0.3 revision A |

To see a list of the device's configuration that are set to explore further configuration options, run the following shell command on a raspberry pi to which a RAK811 is connected:
```sh
rak811v3 get-config lora:status
```

### DevEUI
#### Preconfigured

The RAK811 pHAT has a pre-configured device EUI.
When the device is connected to a raspbery pi, the device EUI can be read by executing the following shell command on the raspberry pi:
```sh
rak811v3 get-config lora:status | grep DevEui
```

#### Manual
The RAK811 pHAT's DevEUI can be changed, in shell or in python, replacing the preconfigured device EUI, and persisting across reboots.

```sh
rak811v3 set-config lora:dev_eui 70XXXXXXXXXXXXXX
```

```python
from rak811.rak811_v3 import Rak811

lora = Rak811()

DEV_EUI = "70XXXXXXXXXXXXXX"
lora.set_config(f'lora:dev_eui:{DEV_EUI}')
```